calliope-mini/microbit-dal #12: Several issues with BLE services

MarkusPiotrowski, 2019-08-22:
Hi,
I'm actually testing some bluetooth services and am experiencing several problems. Some of them are already known.

1. Accelerometer data over BLE does not work
As reported by @smartyw in #7.

2. Callibration of compass not possible if `Bluetooth-IO-Pin-Dienst` is working.
I'm not sure if this is related to #4. However, calibration fails with the last dot of the cirlce and shows several times 'sad face'-020. When not using the `Bluetooth-IO-Pin-Dienst` compass calibration works nicely.

3. The 'compass calibration' service is missing
This bluetooth characteristic (E95DB358-251D-470A-A062-FA1922DFA9A8) is available on the micro:bit. It can be used to invoke the compass calibration. 
The 'problem' is that some libraries, eg. python-bluezero (https://github.com/ukBaz/python-bluezero), are expecting this service, so you cannot use this libraries without manually patching them.

I'm using a 'new' Calliope (name on left side below A button, '1.3' on the back-side right to the USB port).

Regarding topic 1 I think that @smartyw has already gave you some hints where the problem may be located.

Do you think that you will address this issues in the near future? E.g. #7 is from June 2018.

Some response here on GitHub would be nice!

**Set-up to reproduce**
Generate a program with the makecode editor using bluetooth blocks:

[image]

Use "no pairing required" in the project setup:
[image]

You need to remove the `Bluetooth-IO-Pin-Dienst` to make the other things working.

Now, use some software (e.g. Linux blueZ over `bluetoothctl` or an app like Nordic Semiconductor's _nRF Connect_ to analyze the Calliope's bluetooth services.

joernalraun, 2019-08-22:
Hi Markus,
if you have MakeCode (or Open Roberta) running on your own system you can use the updated Version of the DAL: https://github.com/calliope-mini/microbit-dal/tree/merge-upstream
I think that those issues should be fixed.
This one will be the new master in a short while and than it should be distributed via the editors directly.
Best
Joern

MarkusPiotrowski, 2019-08-22:
@joernalraun Thank you! I'll try to set-up a local version of MakeCode and test the updated version of the DAL.

MarkusPiotrowski, 2019-11-13:
I have recognized that `merge-upstream` was merged into the main branch approx. 1 month ago. What are the next steps to make this part of the firmware and/or include the changes to the editors?

woernerth, 2020-02-16:
Any news regarding the ble-accelerometer issue?

joernalraun, 2020-02-16:
> Any news regarding the ble-accelerometer issue?
Yes, indeed! If you try the new Open-Roberta or the current MakeCode beta (MakeCode.calliope.cc/beta) you will have the latest version!

woernerth, 2020-02-19:
Almost everything is working fine. Thank you for the ...beta link.

I think there is a problem running the temperature and accelerometer service at the same time. I can't recieve temperature events in my application if the accelerometer service is running in parallel.