calliope-mini/microbit-dal #17: Serial port cannot be recognized correctly on new calliope windows 7 system

whisky-weisiqi, 2020-11-04:
hi，
I encountered a problem when using the new version of calliope (name on left side below A button, '2.0' on the back-side right to the USB port)
1.When I was using calliope, I plugged into a computer with windows 7 system, the serial port driver would automatically fail to install, resulting in the serial port of my board cannot be read, but it can be installed correctly on windows 10, and the serial port can also be read.
Do you have any good solutions? Thank you！

joernalraun, 2020-11-04:
Hi!
In this case it should not be a problem of the Calliope mini or Calliope mini version 2.
It is a general problem of Windows 7 as the USB drivers have to be installed manually.
There is an info on the mbed site that should help you to install the required driver: https://os.mbed.com/docs/mbed-os/v5.14/tutorials/windows-serial-driver.html
I hope it helps!

whisky-weisiqi, 2020-11-05:
Hi！
I can use calliope mini to install the serial port in windos 7, as shown in the figure
[image]
But when I use calliope mini version 2, the serial port cannot be installed correctly.It will appear that CDC cannot be installed correctly
[image]
Do you have any good solutions? I want to use calliope mini version 2 on windows 7
Thank you！

joernalraun, 2020-11-05:
Mh, maybe you need to use the DAPLink version instead: https://calliope.cc/media/pages/start/tipps/1050954546-1604570158/kl26z_calliope_if_crc.hex
Please tell me if that works better.

whisky-weisiqi, 2020-11-05:
Hi,
I plug the calliope mini version 2 into the computer and use the hex file you gave me. I press and hold the reset button, and then drag the hex file into the disk displayed as mini, and plug the calliope mini version 2 into the computer with windows 7 system. It will automatically Install the driver, the installation effect is as follows:
[image]
The serial port is installed successfully. I am downloading a hex file that uses the serial port to the mini disk, and the serial port can be used normally.
[image]
I plugged in calliope mini version 2, but BBC:microbit is displayed on the arduino, I hope it can display calliope
[image]
There is another problem. I used a calliope mini version 2 to install the serial port driver on the windows 7 system, unplug it and plug it in, the serial port can be recognized. If I am using another new calliope mini verion 2 on a Windows 7 computer with the driver installed, but the serial port cannot be recognized, is there a solution?
[image]
Thank you so much for helping me！

whisky-weisiqi, 2020-11-06:
Hi,
I connected the calliope mini version 2 to the computer of the windows 7 system and used the hex file you sent me. I pressed the reset button, dragged the hex file into the mini disk, and plugged it in again.The system will automatically install the driver, as shown below:
[image]

whisky-weisiqi, 2020-11-06:
I dragged a hex that uses a serial port into the calliope mini version 2 disk, and the serial port can be used normally.But I was recognized as BBC micro:bit on arduino, and I hope it can be recognized as calliope normally.
[image]

whisky-weisiqi, 2020-11-06:
Another problem is that one of my calliope mini version 2 has a serial port driver installed on the windows 7 system, but I use another new calliope mini version 2 to plug it into the computer with the installed windows 7 system, and it is not correct. Identify the serial port. Is there a way to install only one driver on windows 7 and then identify all calliope mini version 2.thank you very much!
[image]